geeekpi/upsplus #115: Raspberry power intermittent when battery is lower than batt_voltage paramater on script

budiroots, 2022-12-07:
Hi dev,

Currently i have more than 10 UPS EP-0136 and running on each Pi 4.
Your python script is working well when power outage happened and battery voltage is lower than we configured, the Pi will shutdown automatically. When power comes in next 3-4 hour, the Pi will turn on automatically with back-to-ac script enabled.

We are facing an issue when power outage happen more than 48H, I check with power meter the battery voltage is draining little by little when Pi is off and power outage still happening, when power comes the raspberry power will intermittent turn on and turn off.

Do you have script to check and charge the battery first when power comes, after the battery voltage is more than configuration the Pi will turn on.

Please help.

Thank you

hellresistor, 2022-12-20:
specs of your power supply ( amp)?

miamilabs, 2022-12-26:
> specs of your power supply ( amp)?

Used multiple power units.

140w Apple USB.
Multi Country plug with 5V & 5A USB A / USB-C to USB-C.

The test unit which i have and other units in my US office have also same symthoms as the user above.

budiroots, 2023-04-26:
> specs of your power supply ( amp)?

I use Samsung fast charging, 5v 3A type-c connector

yoyojacky, 2024-04-08:
When the battery is connected to the device, although the UPS is turned off and does not output voltage, due to the internal resistance of the battery and the characteristics of the chip, the USB interface and TYPE-C interface behind the UPS will still have 3.3V power supply, so if When using UPS, please remove the battery to avoid running out of battery power.

closed 2024-04-08 by yoyojacky