geeekpi/upsplus #117: ina219.DeviceRangeError: Current out of range (overflow), for gain 0.32V, device limit reached

wojciechczyz, 2022-12-16:
When installing software I am getting this error. I have reported it at ina219 library, yet they point back to the board itself.

https://github.com/chrisb2/pi_ina219/issues/32

pi@piraspberry:~/upsplus $ curl -Lso- https://git.io/JLygb | bash
Welcome to 52Pi Technology UPS Plus auto-install Program!.
More information please visit here:.
-----------------------------------------------------.
https://wiki.52pi.com/index.php/UPS_Plus_SKU:_EP-0136.
-----------------------------------------------------.
Start the configuration environment check....
Network status is ok....
Start the software check....
git has been installed..
Installing pi-ina219 library....
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pi-ina219
  Downloading pi_ina219-1.4.0-py2.py3-none-any.whl (9.9 kB)
Collecting mock
  Downloading https://www.piwheels.org/simple/mock/mock-4.0.3-py3-none-any.whl (28 kB)
Collecting Adafruit-GPIO
  Downloading https://www.piwheels.org/simple/adafruit-gpio/Adafruit_GPIO-1.0.3-py3-none-any.whl (38 kB)
Requirement already satisfied: spidev in /usr/lib/python3/dist-packages (from Adafruit-GPIO->pi-ina219) (3.5)
Collecting adafruit-pureio
  Downloading https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-1.1.9-py3-none-any.whl (12 kB)
Installing collected packages: adafruit-pureio, mock, Adafruit-GPIO, pi-ina219
Successfully installed Adafruit-GPIO-1.0.3 adafruit-pureio-1.1.9 mock-4.0.3 pi-ina219-1.4.0
pi-ina219 Installation successful..
Installing smbus2 library....
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting smbus
  Downloading https://www.piwheels.org/simple/smbus/smbus-1.1.post2-cp39-cp39-linux_armv7l.whl (39 kB)
Requirement already satisfied: smbus2 in /home/pi/.local/lib/python3.9/site-packages (0.4.2)
Installing collected packages: smbus
Successfully installed smbus-1.1.post2
smbus2 Installation successful..
create /home/pi/bin directory....
Create python3 script in location: /home/pi/bin/upsPlus.py Successful.
Create UPS Plus IoT customer service python script successful.
Add into general crontab list..
crontab has been created successful!.
------------------------------------------------------------
------Current information of the detected Raspberry Pi------
------------------------------------------------------------
Raspberry Pi Supply Voltage: 5.536 V
Raspberry Pi Current Current Consumption: -2.691 mA
Raspberry Pi Current Power Consumption: 16.821 mW
------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pi/bin/upsPlus.py", line 40, in <module>
    batt_current = ina_batt.current()
  File "/home/pi/.local/lib/python3.9/site-packages/ina219.py", line 196, in current
    self._handle_current_overflow()
  File "/home/pi/.local/lib/python3.9/site-packages/ina219.py", line 246, in _handle_current_overflow
    self._increase_gain()
  File "/home/pi/.local/lib/python3.9/site-packages/ina219.py", line 271, in _increase_gain
    raise DeviceRangeError(self.__GAIN_VOLTS[gain], True)
ina219.DeviceRangeError: Current out of range (overflow), for gain 0.32V, device limit reached
UPS Plus Installation is Incomplete! ... failed!
Please visit wiki for more information:.
-----------------------------------------------------.
https://wiki.52pi.com/index.php/UPS_Plus_SKU:_EP-0136.

toruvinn, 2022-12-28:
That's the output of `upsPlus.py`, do you consistently get that error? I know I get some bus errors when trying to read during battery probing, so perhaps your timing was unfortunate? Granted, the errors are quite different, but with "correct" timing, I don't know what else can break. ;-)

wojciechczyz, 2023-01-22:
This is consistently the same error. Changed batteries, the same error. Tried updating firmware too, still the same error.

yoyojacky, 2024-07-01:
> This is consistently the same error. Changed batteries, the same error. Tried updating firmware too, still the same error.

Could you please execute following command in a terminal ？ 
```bash 
sudo i2cdetect -y 1 
``` 
and send it back to me ?