geeekpi/upsplus #121: Does not recover if batteries are fully discharged

dfcolornet, 2022-12-29:
I have three (3) of these RPi UPS boards, and they all have the same problem: If I shut down the RPi and disconnect the power and don't use it at all for a few weeks, when I reconnect the power there is one (1) blinking blue light every second, and ever third blink it tries to power the RPi on (red LEDs both on the UPS board and on the RPi) for about a second, and then the RPI shuts down, because there is not enough power to run it.  However, no matter how long I leave it plugged on, it does the same thing, and the batteries never charge enough to run the RPi, and it just keeps cycling the RPi on every three (3) seconds, and then turns it off, with one blue light continually blinking.  The only solution I have found is to physically remove the batteries (18650s) from the UPS board, charge them externally, then re-insert them into the UPS board.  Then it starts working again, but it is annoying/difficult to have to take the bottom piece off and charge the batteries externally.  Is there any way that I can leave the batteries in the UPS and get them to charge without removing them and having to charge them externally?

miamilabs, 2022-12-30:
We do expect same issue, there are many posts about this.

We did already try Quck Charge, PD, Apple 140W and Powerbanks and nothing works and developer dont give any answer to this issue.

hellresistor, 2023-01-05:
My method.
Remove baterries , Unmount UPS from Rpi. Put batteries back on ups dettached, and connect power cable to ups. and start charge :| workaround for people dont have extra charger

dfcolornet, 2023-01-06:
Thanks for the suggestion - but that is even more work than removing the batteries and charging them externally (and I have plenty of chargers).  So has anyone seen any response from Geeek Pi on this problem?

Something else that I have noticed ... not as significant, but adds to this problem ... is that leaving the UPS unplugged for a while ... even with the RPi turned "off" ... drains the battery a LOT faster than just leaving the 18650s on a shelf.  I'm not sure whether the UPS board is still trickling current off the batteries when "off"?  This is when the RPi has been shut down and there is no red light on the RPi board itself (which I think means that the RPi should not be drawing any current), and all of the lights on the UPS board are also off (including the blue LEDs).  Not sure if there is any work-around for this, other then (again!) physically removing the batteries?

hellresistor, 2023-02-11:
> 

well i think maybe to much people want this device to an "Big/hard-end" situation..

hmmm.. if you have to much power failures at home/place, maybe you need try solve that problem first, electricity stability. because this is not be to use like an as "Home"-UPS. this is more a away to prevent damaged data, when powerwall failed, device permits raspberry keep runing some time and send order to shutdown OS/Rpi correctly prevent SDcard/data corrupt/damaged.... 
That is my view point of this device utility.

miamilabs, 2023-02-11:
> 
you got a phone number from Putin so i can ask if he stops shoot rockets and i can remove then app UPS and the PI gateway which keeps the gateway online in case fiber and starlink is down.

hellresistor, 2023-02-11:
🤣  lol .. let me see on my phone contacts... nop. 
well maybe + easy to you to call to Zelens to leave the idea of corrupt the old politicals agreements (sovietic times) and crying for help to "nato" xD can i say, my country just send to ukraine all iron garbage in stock ... LOL system cleaning ;) 

well, friend i sorry for your situation. :| i just can deliver you some ideas... try get an car battery, to keep rpi running, and with a battery charger you charge carbattery when electricity avaliable :\

Rikests, 2024-01-02:
Agh... I am facing this exact issue. Got 8 boards after testing one of these for a month. If only there was a way to tell the UPSplus not to power on the PI until battery is charged.
Will have to arrange a return of these boards, cause I cant charge these manually every time powers off for a week.

yoyojacky, 2024-07-01:
When using UPS Plus, the internal resistance of the battery will also consume the battery power when it is not in use. Additionally, because the chip is a DC-to-DC booster, the circuit is always connected once the battery is plugged in, and it will not be completely cut off. If you use it for a long time, it is recommended to connect it to a power supply. When the external power supply is disconnected, the Raspberry Pi will not be instantly powered off; it will have enough time to allow you to save data and shut down. This is the actual use case for UPS, rather than continuously supplying power with the battery until the battery is depleted and then connecting the power supply. The design of UPS includes low-voltage protection, such as shutting down the UPS when the voltage is below 3.7V.

closed 2024-07-01 by yoyojacky

WhatIsThisAllAboutSilly, 2024-12-31:
I've got a unit I've bought in Dec 2024 and it still has this issue.

Is there an updated firmware I can try that fixes it?

Otherwise utterly useless and will be returned...

yoyojacky, 2025-02-06:
> I've got a unit I've bought in Dec 2024 and it still has this issue.
> 
> Is there an updated firmware I can try that fixes it?
> 
> Otherwise utterly useless and will be returned...

Please contact our sales men by sending email to sales@52pi.com

xlla, 2025-06-14:
> Additionally, because the chip is a DC-to-DC booster, the circuit is always connected once the battery is plugged in, and it will not be completely cut off.

> The design of UPS includes low-voltage protection, such as shutting down the UPS when the voltage is below 3.7V.

while it goes into low-voltage protection stage, does it cut off the DC-DC booster ?