geeekpi/upsplus #122: Please make the firmware honour the fixed battery max/min voltage settings

toruvinn, 2023-01-07:
Originally posted in https://github.com/geeekpi/upsplus/issues/87#issuecomment-1366563587
```
$ upsPlus_iot.py 
{
  "PiVccVolt": 5.148,
  "PiIddAmps": 984.861227922624,
  "BatVccVolt": 4.204,
  "BatIddAmps": 80.0,
  "McuVccVolt": 3290,
  "BatPinCVolt": 4198,
  "ChargeTypeCVolt": 9081,
  "ChargeMicroVolt": 64,
  "BatTemperature": 40,
  "BatFullVolt": 4000,
  "BatEmptyVolt": 3600,
  "BatProtectVolt": 3700,
  "SampleTime": 2,
  "AutoPowerOn": 1,
  "OnlineTime": 161237,
  "FullTime": 161237,
  "OneshotTime": 161237,
  "Version": 10
}
```

You can see I've set:
```
  "BatFullVolt": 4000,
  "BatEmptyVolt": 3600,
  "BatProtectVolt": 3700,
```
But unfortunately the UPS seems to ignore it.
Of course I've done `bus.write_byte_data(DEVICE_ADDR, 42, 1)`:
```
FixedVbat register is: 1
Full Voltage register is: 4000
Empty Voltage register is: 3600
```
Firmware version is 10, as stated above. I would really wish to set the max voltage to some arbitrary number, to prolong the battery life (not to mention, I really want the UPS to stop constantly charging them when they are full already), probably at around 60-70% capacity, that's more than enough for my use case.

Would that be possible to make ups+ actually honour those settings, please? Additionally, what are those settings used for, currently? I feel they are completely ignored?

Thank you.

yoyojacky, 2024-04-08:
The battery voltage cannot be set, it can only be collected, that is, the battery voltage is detected, and there is no way to set the battery voltage and current.

closed 2024-04-08 by yoyojacky

toruvinn, 2024-09-28:
Does this mean that register 42/0x2A which is still listed on the wiki was a NOOP from the start?
I cannot find the example scripts that also modify the charging thresholds. I of course understand that modification to any registers won't change the battery parameters, but the request was regarding the ability to change thresholds (i.e. batt voltage when UPS performs shutdown, and voltage when it stops charging the battery).
Additionally documentation states that in V9:

> Version V9:
>  - [Bug fix] The manual voltage can not be set, please note that the manual setting threshold voltage only affects the charging logic, and does not change the parameters of the battery itself.

Which is basically what I meant above too. For me, modification of the thresholds does nothing. I would absolutely love to set the shutdown voltage a bit higher (to prevent battery from discharging below 30-40%) and max voltage (to stop charging) around 70% of the battery capacity to prevent unnecessary wear. This is what that issue was about, apologies for not being very clear.