geeekpi/upsplus #123: PCB burnt

HS-Tech-kr, 2023-01-16:
PCB burnt when I replace the battery.

[image]
[image]
[image]

zaretajoesin, 2023-01-17:
Forwarded from the Technical Support Team:

This is a frequently asked question (FAQ), it usually contains the following possible causes, please check them one by one, if you need to repair, please contact the after-sales support team.

1) Reverse battery installation.
2) Static electricity in excess of 3KV on the hands, exceeding the tolerance limit.
3) The battery was installed without the USB charging cable removed.

All our products are fully tested before leaving the factory and repairs are usually free of charge, however you cannot contact the after sales team here, please use the purchase channel to contact the after sales team.

HS-Tech-kr, 2023-01-17:
first, I installed my old 18650 battery, but it hasn't enough current capacity. (but UPS plus worked well)
so i decided to buy new battery from reliable company such as picture above.
As soon as install the battery, I could smell like burn.
After that, there is no 0x17.

[image]

ArjenR49, 2023-03-12:
The batteries are connected in parallel. Obviously the connection between the positive and negative terminals of the two batteries is made using traces on the PCB. When the two batteries have different voltages, like when one is fully charged and the other is half charged, as it comes fresh from the shop, there will be a very large equalizing current between the batteries once they are seated in the UPS's battery holder. The fully charged one will charge the other battery and the current from one battery to the other will be very large until the voltages become equal. The current could be as much as 10 A or perhaps even more depending on the type of battery. The burnt trace on the PCB is obviously not designed for this situation.

It would be prudent of the manufacturer to warn of mixing batteries with different terminal voltages; I don't know if there actually is such a warning, however it could be regarded as common knowledge.

Lithium batteries can cause dangerous situations when handled wrongly. After all it is for good reasons that packing such batteries without protection in airplane cargo is prohibited.
Speaking of protection, Li batteries with a protection circuit built in (actually added onto the terminal end) are too long for the UPS's battery holder, so there you are ...

yoyojacky, 2024-04-08:
Please contact with sale man and ask for replacement.

closed 2024-04-08 by yoyojacky

HS-Tech-kr, 2024-04-09:
> Please contact with sale man and ask for replacement.

Thank you for your support.
I didn't expect this kind of warranty service.
The seller said he will send the replacement tomorrow.
I am very happy.

yoyojacky, 2024-04-10:
> > Please contact with sale man and ask for replacement.
> 
> Thank you for your support. I didn't expect this kind of warranty service. The seller said he will send the replacement tomorrow. I am very happy.
OK, have a nice day!

Leorivasc, 2024-05-29:
Hi all. I'd like to know if this is a 'typical' failure. I've had mine for like 6 months, never managed to make it work and only now found that mine had the same burnt place. Is it worth replacing? Would hate to buy again and burn it without knowing. Thanks.

CosmicFurFace, 2024-05-30:
From section 4.1 of the wiki (https://wiki.52pi.com/index.php/EP-0136#Specifications):
`Note: Different types of batteries cannot be mixed!`

My understanding is that it's "common knowledge" (often expressly noted) that, when using multiple Li-ion cells, one should:
- *Never* mix cell brands, capacities, or ages.
- *Always* ensure all cells are equally charged/discharged.

Treating Li-ion cells like AA batteries ("just stick a couple in there") is, to my understanding, dangerous, and I'd expect the behavior described ("decided to buy new battery" and pair with old one) to produce excessive currents in most equipment I've used. 

Even if the circuit is protected against unbalanced parallel cells, I've read that [unbalanced] charging can still be a problem.

yoyojacky, 2024-06-06:
> Hi all. I'd like to know if this is a 'typical' failure. I've had mine for like 6 months, never managed to make it work and only now found that mine had the same burnt place. Is it worth replacing? Would hate to buy again and burn it without knowing. Thanks.

Could you please send me a picture about  the burned place?

yoyojacky, 2024-06-06:
> From section 4.1 of the wiki (https://wiki.52pi.com/index.php/EP-0136#Specifications): `Note: Different types of batteries cannot be mixed!`
>
> My understanding is that it's "common knowledge" (often [...]

yes, that's correct, li-po batteries are different from AAA batteries.

Leorivasc, 2024-06-06:
Hi there, thanks for your answer. Well, I can't send pictures because out of frustration, once I realized it was burned, I threw the board to the trash and don't have it any more. In any case, it was the exact same situation as the picture above, same model, burned track at the same exact place. I even tried to bridge the burnt track using a screwdriver, and it smoked. The main symptom was that it threw only 3.7V out of the USB and the pins going to the Pi, so it didn't power up. Batteries seemed to charge well when connected, the blue lights worked as intended.
I bought the card with the [GeekPi case](https://www.amazon.com/dp/B09M8LSKZ8?psc=1&ref=ppx_yo2ov_dt_b_product_details) and the frustration came from the message "DO NOT PLUG Power supply to Raspberry Pi in USB-C port during Using UPS, it may damage UPS!!!! " that is found oh [this website](https://wiki.52pi.com/index.php?title=EP-0136). Well, thing is, the case design don't allow to mount the UPS to the Pi and THEN insert the batteries as the wiki instructs, so I thought it was a combination of bad design from the case manufacturer or the board itself for having that somewhat obvious weakness that I may have inadvertently burnt on the several attempts to make it work.. or was it because I mounted it with the batteries on? Will never know.
Anyway, thanks for asking.

yoyojacky, 2024-06-06:
Could you please send me your order number ? I can arrange my teammate follow up this case.

Leorivasc, 2024-06-06:
Sorry I don't get it, what do you mean?

yoyojacky, 2024-06-06:
I mean, you can send me your order number of the UPS, i will arrange my teammate to follow up the case, they may send you replacement if they checked the order number and they will contact with you ASAP.

Leorivasc, 2024-06-06:
Ow, thanks. This is it. Order Placed: December 26, 2023, Amazon.com order number: 112-1284073-0034610.
Thanks very much for your consideration