geeekpi/upsplus #124: Power Adapter Report Online and Offline via Online Diagnostics

SpekOregonian, 2023-01-25:
Running diagnostics on my ups (https://api.52pi.com/), following messages received;

Power adapter offline.

It seems that your battery is still not fully charged.

You are using a fast charger to charge through the USB-C port.

It is recommended to use a fast charging charger.

I am also experiencing stats freezing like battery percentage and uptime (all three counters). I have re-installed and reseated and experience same issues.

I am running firmware 14 per the registers

Hardware: Raspberry Pi 4b
OS: Debian 11 (bullseye)
Kernel: 5.15.84-v8+
Firmware: 14

SpekOregonian, 2023-01-25:
Per the time counters, between 26 and 30 minutes from power up the counter stops incrementing.

SpekOregonian, 2023-01-25:
Output from full featured script The diagnostic page is reporting just the following now;

You are using a fast charger to charge through the USB-C port.

However, Battery Percentage and all timers are frozen. They did increment for a period.

python upsplus/Full-featured-demo-code.py 
Raspberry Pi power supply voltage: 5.168 V
Current current consumption of Raspberry Pi: 1245.416 mA
Current power consumption of Raspberry Pi: 6442.389 mW
Batteries Voltage: 4.204 V
Battery current (charging), rate: 8.049 mA
Current battery power supplement: 175.610 mW
Current processor voltage: 3306 mV
Current Raspberry Pi report voltage: 5218 mV
Current battery port report voltage: 4206 mV
Current charging interface report voltage (Type C): 8851 mV
Current charging interface report voltage (Micro USB): 54 mV
Currently charging through Type C.
Current battery temperature (estimated): 43 degC
Full battery voltage: 4274 mV
Battery empty voltage: 3780 mV
Battery protection voltage: 3500 mV
Battery remaining capacity: 92 %
Sampling period: 2 Min
Current power state: normal
No shutdown countdown!
Automatically turn on when there is external power supply!
No restart countdown!
Accumulated running time: 1821 sec
Accumulated charged time: 1829 sec
This running time: 1821 sec
Version number: 14

SpekOregonian, 2023-01-25:
Reboot after disconnecting the USB-C power;

pi@spek-rsp003:~ $ python upsplus/Full-featured-demo-code.py 
Raspberry Pi power supply voltage: 5.068 V
Current current consumption of Raspberry Pi: 1270.311 mA
Current power consumption of Raspberry Pi: 6344.828 mW
Batteries Voltage: 4.144 V
Battery current (discharge), rate: -2060.000 mA
Current battery power consumption: 10887.805 mW
Current processor voltage: 3306 mV
Current Raspberry Pi report voltage: 5218 mV
Current battery port report voltage: 4206 mV
Current charging interface report voltage (Type C): 8851 mV
Current charging interface report voltage (Micro USB): 54 mV
Currently charging through Type C.
Current battery temperature (estimated): 43 degC
Full battery voltage: 4274 mV
Battery empty voltage: 3780 mV
Battery protection voltage: 3700 mV
Battery remaining capacity: 92 %
Sampling period: 2 Min
Current power state: normal
Shutdown countdown: 240 sec
Automatically turn on when there is external power supply!
Restart countdown: 10 sec
Accumulated running time: 1821 sec
Accumulated charged time: 1829 sec
This running time: 1821 sec
Version number: 14

yoyojacky, 2024-04-08:
Could you please try to reboot Raspberry Pi and execute this python script again? it seems like the I2C communication has been interrupted ...

closed 2024-07-01 by yoyojacky