geeekpi/upsplus #127: Blue LEDs stopped indicating actual battery charge

ArjenR49, 2023-03-12:
After several months away from the UPSplus and the Pi attached to it, a few days ago I did a f/w upgrade. The info on the Github pages about the current f/w version seems no longer there, so I didn't know what to expect. The UPS upgraded to f/w version 14. So far so good. However, no info about any changes or improvements, as far as I could see.

What I noticed myself is that now the blue LEDs behave erratically. Only the first two are lit. Number 2 is blinking.
The UPS has been on its charger all the time since the upgrade and the batteries are fully charged as per the UPS registers and a digital voltmeter connected to the battery terminals. It's just the LEDs not reflecting the situation correctly.
I recall having read a comment about the LEDs misbehaving, but I cannot find it now.

Is there a way to make the LEDs behave again as they used to do, in a sensible way?

frtz13, 2023-04-13:
same here, for several months now. currently, all blue LEDs are off.

ArjenR49, 2023-04-14:
When I returned to my cottage where the UPS Plus was in use, I decommisioned the UPS Plus despite all the effort I have put in it over the last year. I had lost control of it months before that when I was away.

Years ago when I had just started with Raspberry PI's I bought a PiJuice UPS which I tested and discussed about with responsive developers and voluntaries at the time, then in following years more or less forgot about.
Compared to the several other brand Pi UPS's I have, the Pi Juice proved the easiest and best supported.

I totally reinstalled the Pi that had the UPS Plus. That removed also the last trace of HA, which had proved a nuisance and difficult to uninstall. It kept looking up some domain address.
The PiJuice doesn't have the extra cooling from the Ice Tower, but it hasn't been a problem.
At least I can stop worrying about the UPS Plus and it's Lithium cells.

galtobellojr, 2023-04-14:
I've written off the UPS Plus for some time now, and have a couple of Red Reactors obtained via Kickstarter that I'll be trying next. 

It's a shame, but I concluded some time ago that the serious battery drain issue is a show-stopper with no workaround. Never mind other ongoing questions regarding support and engineering.

Good luck on your search.

yoyojacky, 2024-04-08:
When the battery is not used for a long time, its internal resistance will also consume battery power. Additionally, due to chip limitations, the UPS cannot completely cut off power. Therefore, please remove the battery when the UPS is not in use for an extended period of time.

closed 2024-04-08 by yoyojacky