geeekpi/upsplus #130: OSError: [Errno 121] Remote I/O error

p1r473, 2023-04-29:
Usually, the application works for me.
However some times, I get the following error:

```
------------------------------------------------------------
-------------------Batteries information-------------------
------------------------------------------------------------
Voltage of Batteries: 3.560 V
Battery Current (discharge) Rate: -2283.902 mA
Current Battery Power Consumption: 8024.390 mW
------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pi/upsplus/upsplus.py", line 66, in <module>
    aReceiveBuf.append(bus.read_byte_data(DEVICE_ADDR, i))
  File "/usr/local/lib/python3.9/dist-packages/smbus2/smbus2.py", line 433, in read_byte_data
    ioctl(self.fd, I2C_SMBUS, msg)
OSError: [Errno 121] Remote I/O error                  
```

botboe, 2023-06-13:
I can confirm this behavior.

dacarson, 2023-07-12:
When I try to do an OTA update, I do get the same error:
```
pi@raspberrypi:~/bin/upsplus $ python3 OTA_firmware_upgrade.py 
Traceback (most recent call last):
  File "/home/pi/bin/upsplus/OTA_firmware_upgrade.py", line 23, in <module>
    aReceiveBuf.append(bus.read_byte_data(DEVICE_ADDR, i))
  File "/home/pi/.local/lib/python3.9/site-packages/smbus2/smbus2.py", line 433, in read_byte_data
    ioctl(self.fd, I2C_SMBUS, msg)
OSError: [Errno 121] Remote I/O error
```
However, other commands work:
```
raspberrypi:~/bin $ python3 upsPlus.py 
------------------------------------------------------------
------Current information of the detected Raspberry Pi------
------------------------------------------------------------
Raspberry Pi Supply Voltage: 4.936 V
Raspberry Pi Current Current Consumption: 1012.447 mA
Raspberry Pi Current Power Consumption: 5076.535 mW
------------------------------------------------------------
-------------------Batteries information-------------------
------------------------------------------------------------
Voltage of Batteries: 4.200 V
Battery Current (Charging) Rate: 778.049 mA
Current Battery Power Supplement: 3278.049 mW
Successfully set the protection voltage to: 3700 mV
------------------------------------------------------------
Currently charging via Micro USB Port.

dacarson, 2023-07-13:
I have not seen this error once I closed the bus when I finished with it. That is, adding:
`bus.close()`
to the end of the python script (or where it exited) addressed the issue for me.

sfatula, 2023-08-09:
To do the firmware update, are you putting the UPS into firmware update mode? That error indicates it is not in the mode.

sfatula, 2023-08-09:
Hmm, I was speaking of your post while doing python3 OTA_firmware_upgrade.py 

That requires putting the UPS in firmware update mode first was my only point. That is not run via cron.

dacarson, 2023-08-09:
For me, I had run out of file handles when I tried to use `python3 OTA_firmware_upgrade.py`

p1r473, 2024-02-05:
Only fix that worked for me was putting it in a loop that attempts the operation 3 times, because sometimes it just fails.