geeekpi/upsplus #132: Make Firmware Public

Rogash1, 2023-07-24:
I own a UPSPlus for a while now and face several issues with it. I really want to like the product, but the firmware behaves very strangely and a lot of functionality is not appropriately documented. Can you give us the source code of the main microcontroller so the community is able to fix the issues?

Rikests, 2024-03-07:
This could be the saving grace for this device. I believe people could fix up some basic issues or add use cases them selves. If only the source of firmware of the UPS plus controller was available.

yoyojacky, 2024-04-08:
We're not reluctant to release the firmware source code. It's because this involves power management, and we're concerned that users might modify the firmware privately, leading to device damage or even fire hazards. That's why it's not open to the public. However, feel free to raise any issues you encounter, and we'll see if we can assist you in resolving them.

Rikests, 2024-06-26:
The problem is that the product is unusable for some, because of this power issue https://github.com/geeekpi/upsplus/issues/121.
If you would implement a function where the battery would be charged to a reasonable voltage before powering on the raspberry via BACK-TO-AC function I would not have any reason to touch the firmware of UPS plus.
I really like the product, just this software issue makes it unusable and 8 UPSplus boards are just gathering dust in my shelf.

yoyojacky, 2024-07-01:
> The problem is that the product is unusable for some, because of this power issue #121. If you would implement a function where the battery would be charged to a reasonable voltage before powering [...]

Could you please  send E-mail to : jacky.li@52pi.com  and we will help you resolve this problem.

dacarson, 2024-07-16:
I have the same problem. If the batteries get too low, it won't turn back on. I have left the unit sitting for days, and it doesn't power up again as described in #121 . I end up having to do the exact steps that this user reported: https://github.com/geeekpi/upsplus/issues/121#issuecomment-1371936066 , ie unscrew and unmount the UPSPlus board from the RPi, charge up the batteries, then mount it back on again.

If there was a firmware setting that did something like, power up RPi when `Battery Remaining` reached X %, rather than trying to power up immediately, that would solve the problem.

Should I also email the address above too?

yoyojacky, 2024-07-20:
> I have the same problem. If the batteries get too low, it won't turn back on. I have left the unit sitting for days, and it doesn't power up again as described in #121 . I end up having to do the exact steps that this user reported: [#121 (comment)](https://github.com/geeekpi/upsplus/issues/121#issuecomment-1371936066) , ie unscrew and unmount the UPSPlus board from the RPi, charge up the batteries, then mount it back on again.
> 
> If there was a firmware setting that did something like, power up RPi when `Battery Remaining` reached X %, rather than trying to power up immediately, that would solve the problem.
> 
> Should I also email the address above too?

yes, please.

dacarson, 2024-07-29:
I would still like to request making the Firmware public. This would allow the community to implement:

> a function where the battery would be charged to a reasonable voltage before powering on the raspberry via BACK-TO-AC function ...
-- https://github.com/geeekpi/upsplus/issues/132#issuecomment-2191712930

Because the BACK-TO-AC the way it stands today doesn't seem to work for a lot of people, including me.

toruvinn, 2024-09-28:
+1 that would also allow me to fix/workaround my problems that prevent me from using it (i.e. it never stops charging, for whatever reason the manual threshold modification does nothing and therefore looking into its behaviour would allow me to likely fix the issue).

Heck, even a simple ability to send a "stop charging"/"start charging" request to the UPS would let me just workaround the problem.
I really, really could use a functional UPS for my rpi(s), but unfortunately so far have found none without issues that are showstoppers this way or another.

dacarson, 2026-01-31:
This product is now EOL. I have been able to build my own version of the firmware with the features that I needed and solid behaviour. Check it out here: https://github.com/dacarson/UPSPlus_oss