geeekpi/upsplus #133: Can't revert firmware

Rogash1, 2023-07-24:
It seems that the API does not allow to choose a specific firmware version anymore.

In the "OTA_firmware_upgrade.py" script a comment indicates that you can revert the UPS firmware to a specific version:

# You can also specify your version, so you can rollback/forward to the specified version
# r = requests.post(UPDATE_URL, data={"UID0":UID0, "UID1":UID1, "UID2":UID2, "ver":7})_

This post request itself works and returns a download URL for "firmware_v7.bin", but as soon as the script wants to download the firmware a 404 error is returned :

print('Pass the authentication, downloading the latest firmware...')
req = requests.get(r['url']) #<-- here you get the 404 error

isjfk, 2023-12-25:
Seems the support has been dead for at least 2 years.

yoyojacky, 2024-04-08:
please check if the network work properly, it works for me, and i have already upgraded to v10 .