geeekpi/upsplus #135: UPSPLUS is stuck at address 0x18

Jambo52, 2023-08-21:
I have tried to install ups several times, but the address 0x18 is always there. I can't get to 0x17. I removed the batteries and power and tried the OTA update but that doesn't work either. What can I do?

KrzePikPL, 2023-11-21:
I have the same problem with version v10, after startup it works for about 1 minute, then to turn on the device I have to manually remove the battery but with no improvement, after restoring the power the same problem persists

Jambo52, 2023-11-21:
I didn't solve it either. I switched because the support doesn't work here either. I have now connected the board EP-0136. This does shutdown and starts even after a power failure. Unfortunately I can't help you either.

KrzePikPL, 2023-11-21:
I don't know what board would work, so that it could check the charging status and start automatically, it's a simple script, and the second board I have has problems with it. And the spontaneous turning off of the board disqualifies it because then it turns on itself, but at an unspecified time it is possible that it is a problem with the switching on times because the board has an RTC, so maybe you need to look for it there

KrzePikPL, 2023-11-26:
I wonder if if I used the X728 software (https://wiki.geekworm.com/X728#X728_V2.5), whether the device would accept it and start working with these functions. Because there is clearly a problem with the clock on the original board

yoyojacky, 2024-04-08:
No, it is not fit for X728 software, it just fit for this repo： https://github.com/geeekpi/upsplus.git

closed 2024-04-08 by yoyojacky