geeekpi/upsplus #142: NUT driver for UPSPlus

dacarson, 2024-04-03:
I wrote a NUT driver for UPSPlus (EP-0136), and I would appreciate any feedback before I submit a PR to the NUT repo.

Make sure that you have NUT installed:
`$ sudo apt-get install nut`

**To build it:**
You can clone my forked repository from here:
`$ git clone https://github.com/dacarson/nut`
`$ cd nut`

Generate the configuration files:
`$ ./autogen.sh` 

Generate the Makefiles for the items we need to build and their path
`$ ./configure --with-linux_i2c --with-statepath=/run/nut --sysconfdir=/etc/nut --with-user=nut --with-group=nut --with-pidpath=/run/nut`
parameters meanings
`--with_linux_i2c` Build the NUT i2c linux UPS drivers
`--with-statepath` & `--with-pidpath` Location to put shared file & the PID file (_default location for bullseye is /run/nut_)
`--sysconfdir` Location of the NUT configuration files (_default location for bullseye is /etc/nut_)
`--with-user & --with-group` Username and Group to run the driver as (_default for bullseye is nut:nut_)

Build all the components
`$ make all`

**Test the driver to make sure that it will run ok**
`$ sudo ./drivers/upsplus -DDD -s pi -x port=default -d 1`

If it tests out ok, then copy it to the right directory for bullseye:
`$ sudo cp upsplus /usr/lib/nut`

Add this section to your `/etc/nut/ups.conf`:
```
[pi]
    driver = upsplus
    port = /dev/i2c-1
    desc = "RaspberryPi UPSPlus"
```

**Make sure that the NUT upsplus driver can access the i2c-1 bus:**
`$ sudo chmod a+rw /dev/i2c-1`
(The default user/group for a nut installations (user: nut, group: nut) doesn't have read-write access as the i2c bus device is created by default as user and group read-writable that nut is not a part of. This is something I need to work out how to fix because this is reset when the RPi reboots.)

Now restart your NUT driver
`$ sudo systemctl restart nut-driver`

Test that it is running correctly:
`$ upsc pi`

yoyojacky, 2024-07-01:
Hi dacarson, 
Thanks for your effort, I will arrange our enginner to test it out, and give you a feed back, that looks nice!

dacarson, 2025-08-31:
Just updated the NUT driver to be far more efficient (ie consume less CPU) in getting status over I2C. Incremented the revision to 1.1