geeekpi/upsplus #151: Unit of shutdown delay is not second.

xlla, 2025-06-14:
I'v write a test script to study ups auto shutdown feature. 
It write 120 into ups mcu reg 24, then sleep 10s, then read from reg 24, it was **119** instead of **110** ,  so the unit of shutdown delay seems 10 second,  in other words, decrease by 1 every 10s.

my test script

```
print("set ups power shutdown delay to 120s")
ups.shutdown_delay = 120
time.sleep(10)
#the unit seems is 10s 
#check value
print(f'current ups shutdown delay {ups.shutdown_delay} s')
ups.shutdown_delay = 0

```

 the result log

```
set ups power shutdown delay to 120s
10499-MainThread 2025-06-15 02:29:53,298 upsPlusHelper.py:100 write_ups    DEBUG - write to ups mcu addr 23, reg 24, value 120, length 1
current ups shutdown delay 119 s
10499-MainThread 2025-06-15 02:30:03,414 upsPlusHelper.py:100 write_ups    DEBUG - write to ups mcu addr 23, reg 24, value 0, length 1

```

xlla, 2025-06-14:
To verify my guess, I launch another test, it loops 10 times and print more detailed log.
my guess was wrong, it seems the unit is random.

new test script

```
print("set ups power shutdown delay to 120s")
ups.shutdown_delay = 120
for i in range(10):
    time.sleep(10)
    log.debug(f'current ups shutdown delay {ups.shutdown_delay} s')
ups.shutdown_delay = 0
```

the result log

```
set ups power shutdown delay to 120s
11414-MainThread 2025-06-15 03:09:39,703 upsPlusHelper.py:101 write_ups    DEBUG - write to ups mcu addr 23, reg 24, value 120, length 1
11414-MainThread 2025-06-15 03:09:49,816 upsPlusHelper.py:97 read_ups     DEBUG - read from ups mcu addr 23, reg 24, value 120, length 1
11414-MainThread 2025-06-15 03:09:49,817 test-upsHelper.py:107 <module>     DEBUG - current ups shutdown delay 120 s
11414-MainThread 2025-06-15 03:09:59,830 upsPlusHelper.py:97 read_ups     DEBUG - read from ups mcu addr 23, reg 24, value 120, length 1
11414-MainThread 2025-06-15 03:09:59,831 test-upsHelper.py:107 <module>     DEBUG - current ups shutdown delay 120 s
11414-MainThread 2025-06-15 03:10:09,836 upsPlusHelper.py:97 read_ups     DEBUG - read from ups mcu addr 23, reg 24, value 119, length 1
11414-MainThread 2025-06-15 03:10:09,837 test-upsHelper.py:107 <module>     DEBUG - current ups shutdown delay 119 s
11414-MainThread 2025-06-15 03:10:19,849 upsPlusHelper.py:97 read_ups     DEBUG - read from ups mcu addr 23, reg 24, value 117, length 1
11414-MainThread 2025-06-15 03:10:19,850 test-upsHelper.py:107 <module>     DEBUG - current ups shutdown delay 117 s
11414-MainThread 2025-06-15 03:10:29,863 upsPlusHelper.py:97 read_ups     DEBUG - read from ups mcu addr 23, reg 24, value 117, length 1
11414-MainThread 2025-06-15 03:10:29,864 test-upsHelper.py:107 <module>     DEBUG - current ups shutdown delay 117 s
11414-MainThread 2025-06-15 03:10:39,876 upsPlusHelper.py:97 read_ups     DEBUG - read from ups mcu addr 23, reg 24, value 117, length 1
11414-MainThread 2025-06-15 03:10:39,877 test-upsHelper.py:107 <module>     DEBUG - current ups shutdown delay 117 s
11414-MainThread 2025-06-15 03:10:49,889 upsPlusHelper.py:97 read_ups     DEBUG - read from ups mcu addr 23, reg 24, value 117, length 1
11414-MainThread 2025-06-15 03:10:49,890 test-upsHelper.py:107 <module>     DEBUG - current ups shutdown delay 117 s
11414-MainThread 2025-06-15 03:10:59,898 upsPlusHelper.py:97 read_ups     DEBUG - read from ups mcu addr 23, reg 24, value 116, length 1
11414-MainThread 2025-06-15 03:10:59,899 test-upsHelper.py:107 <module>     DEBUG - current ups shutdown delay 116 s
11414-MainThread 2025-06-15 03:11:09,910 upsPlusHelper.py:97 read_ups     DEBUG - read from ups mcu addr 23, reg 24, value 116, length 1
11414-MainThread 2025-06-15 03:11:09,911 test-upsHelper.py:107 <module>     DEBUG - current ups shutdown delay 116 s
11414-MainThread 2025-06-15 03:11:19,917 upsPlusHelper.py:97 read_ups     DEBUG - read from ups mcu addr 23, reg 24, value 116, length 1
11414-MainThread 2025-06-15 03:11:19,918 test-upsHelper.py:107 <module>     DEBUG - current ups shutdown delay 116 s
11414-MainThread 2025-06-15 03:11:19,919 upsPlusHelper.py:101 write_ups    DEBUG - write to ups mcu addr 23, reg 24, value 0, length 1
```

it costs 20+x seconds from 120 to 119.
it costs less then 10s from 119 to 117, so we missing 118.
it costs 30+x seconds from 117 to 116.

xlla, 2025-06-14:
I found some user was encounter the same problem, in this issue #75 ,
[ArjenR49](https://github.com/ArjenR49) wrote 

> it sets the countdown timer 0x18 register to 60 and then sends a shutdown command to the Pi.

> a shutdown at 3.38 a.m. However the Pi kept running past the shutdown command somehow as can be seen from the syslog until at 4.17 a.m.

about 39 min to cut off the ups after set countdown timer to 60.
it is about average 39 seconds to decrease by 1 to countdown timer.

dacarson, 2026-01-31:
This product is now EOL. I have been able to build my own version of the firmware with the features that I needed and solid behaviour. I believe I have addressed this issue with refactoring the interrupts and timing.
Check it out here: https://github.com/dacarson/UPSPlus_oss