geeekpi/upsplus #152: Firmware 14 stops reporting data

dacarson, 2025-09-09:
I recently updated from Firmware 10 to Firmware 14. Firmware 10 was mostly working fine, though I wanted to test my NUT UPS driver with the newer firmware.
I am running a script to read the data from the i2c bus every 10 seconds to see which fields are actually updated by the sample interval. The script was working fine with FW 10. 
FW 14 runs for a while then stops reporting new data. I can read the registers ok, but they stop updating. I found that I have to reboot the device to get it running again (remove power and batteries, then re-insert power and batteries). When I do so, I see that the Running Time counters are all reset to zero. 
I can set values in the registers but they don't stick. 

I tried downloading firmware 10, but it says the firmware version not found.

dacarson, 2025-09-10:
Thanks for all your insights. I had no issues with FW 10, I never experienced this crash. I wish they made FW 10 available so I could roll back to that version.
The way the device functions, it looks the same as when you put it into "OTA mode" via registers, ie 
`i2cset -y 1 0x17 50 127 b`
I put it in quotes as in my experience, it doesn't enable 0x18. I had to do the other method to update the firmware.
I wouldn't be surprised if it was somehow falling into that state. Everything continues to work "fine" and charges the battery, powers the RPi, but values stop updating.

Can you give me more details on the transistor path?

FWIW I have also asked for the software to be made open source, I emailed them directly. The issue I was facing with FW 10 and I suspect the same with FW 14, was that it powers the RPi back on too soon, it draws too much power and it shuts it down again.
It turns OFF the power to the 5V pin when battery_voltage < protection_voltage, but it turns it back ON when Battery_voltage + 100mV > protection_voltage. The RPi causes more than 100mV drop on the battery while powering on :-( I request to make the 100mV configurable, and I was told no. I should just not use the auto-restart flag.

markkkx000, 2025-11-13:
I am really regretting updating the firmware... I wish i saw this sooner. I hope someone in geeekpi becomes reasonable and make the firmware open-source.

dacarson, 2025-11-13:
They are now working on a new UPS, UPSv6:
https://github.com/geeekpi/upsv6_pub
https://wiki.52pi.com/index.php?title=EP-0250#Description

Their Wiki says that the firmware is customizable. I posted a question about that:
https://github.com/geeekpi/upsv6_pub/issues/14

markkkx000, 2025-11-17:
> If you leave the system running and regularly send requests over I²C, the firmware eventually crashes

@NightFlowed did you confirm whether it happens on both I²C reads and write requests? i was running a script before that just reads battery percentage every 60 seconds and it doesn't freeze the firmware (although i only experimented for about 12 hrs). I only noticed mine crashing after i enabled the crontab tasks that regularly executes the scripts in $HOME/bin (upsPlus.py and upsPlut_iot.py). Those scripts have 'bus.write_byte_data()' lines so i suspect that it only crashes on constant I²C writes. If this is the case, I could still potentially salvage the automatic shutdown function by using a script that just reads the battery voltage and executes a shutdown command if voltage drops to a certain point, so instead of writing to the registers like this:
```
# Reset Protect voltage
bus.write_byte_data(DEVICE_ADDR, 17, PROTECT_VOLT & 0xFF)
bus.write_byte_data(DEVICE_ADDR, 18, (PROTECT_VOLT >> 8)& 0xFF)
print("Successfully set the protection voltage to: %d mV" % PROTECT_VOLT)
```
we just handle it all on the raspberry pi, for the BACK-TO-AC feature, we could just set it to 1 right before the pi executes shutdown... this works under the assumption that regularly reading from the registers is safe and preventing firmware freezes by minimizing the write operations. this eliminates a lot of the possible use cases of this UPS but if i just want the automatic shutdown and BACK-TO-AC function this could work. i'll experiment again for longer durations and see if it still crashes on purely read operations.

deHarro, 2025-11-17:
Hi @dacarson!

The up-to-date version of the wiki for UPSv6 is [EP-0245](https://wiki.52pi.com/index.php?title=EP-0245).

yoyojacky, 2025-11-18:
> [@NightFlowed](https://github.com/NightFlowed) did you confirm whether it happens on both I²C reads and write requests? i was running a script before that just reads battery percentage every 60 [...]

yes, that  is demo code, just for you to reference, everybody has their own requirements, so everybody can modify the python script s to fit for their own requirements.   thus, your new script is functional and great jobs ~ have a nice day~

yoyojacky, 2025-11-18:
> Unfortunately I'm not home at the moment otherwise I would share my script that I use to monitor the UPSPlus. It does not write to i2c but just reads the registers every second (I was trying to work out which registers are updated on the 'sample' interval). The device does stop reporting new numbers after a while. I have not been able to work out the interval - sometimes it stops within minutes other times it stops after a day or so. I have only experienced the issue with the newest firmware (14). The other unit that I have firmware 10 on does not stop. I don't know the difference between 10 and 14.

The UPS functionality was frozen in V9. Subsequent updates have only included safety fixes, such as corrections to battery parameters, rather than functional enhancements. If the microcontroller enters a locked state, it is necessary to check the input power supply and battery parameters. Do not mix batteries from different brands. Additionally, check the internal resistance of the batteries. For the input power supply, verify whether there is excessive ripple. Ensure that the PogoPin positions maintain good contact. On the system level, check if the I2C device files are being occupied by other applications, or update to the latest official system.

markkkx000, 2025-11-20:
i am using this script now which i run using systemd service unit and timer (1min intervals) instead of cron:
```
#!/bin/bash

batt_raw=$(i2cget -y 1 0x17 0x13 w)
usb_c_raw=$(i2cget -y 1 0x17 0x07 w)
micro_usb_raw=$(i2cget -y 1 0x17 0x09 w)

usb_c_volt=$((usb_c_raw))
micro_usb_volt=$((micro_usb_raw))
batt_percentage=$((batt_raw))
batt_cutoff=20		# battery percentage where Pi will execute safe shutdown
power_off_cd=20		# countdown in seconds before UPS cuts power to Pi

# if 0x13-0x14 registers returned 0x0000, treat it as failed i2c read.
if (( batt_percentage == 0 )); then
	echo "ERROR: invalid battery percentage (0%). Skipping..."
	exit 1
fi

echo -n "UPS Batt: $batt_percentage% "

charging=0
if (( usb_c_volt > 4000 )); then
	charging=1
	charging_port="USB-C"
elif (( micro_usb_volt > 4000 )); then
	charging=1
	charging_port="Micro-USB"
fi

if (( batt_percentage <= batt_cutoff )) && (( charging == 0 )); then
	echo "-> UPS low battery, shutting down now..."
	i2cset -y 1 0x17 0x18 $power_off_cd
	echo "UPS will power off in $power_off_cd seconds"
	sleep 2 	# delay shutdown to allow logging to finish
	sudo shutdown -h now
elif (( charging == 1 )); then
	echo "(currently charging via $charging_port)"
else
	echo "(discharging... will shutdown at $batt_cutoff%)"
fi
```
it's been running for about 3 days now, the board seems fine and responsive.

dacarson, 2025-12-05:
> They are now working on a new UPS, UPSv6: https://github.com/geeekpi/upsv6_pub https://wiki.52pi.com/index.php?title=EP-0250#Description
> 
> Their Wiki says that the firmware is customizable. I posted a question about that: [geeekpi/upsv6_pub#14](https://github.com/geeekpi/upsv6_pub/issues/14)

They have updated their site (and my bug report) to say that customization will **not** be possible, surprise surprise. Though it may be possible in the future...